KumarRobotics/kr_autonomous_flight #139: Differentiate 2D and 3D planner in local replan server

fcladera, 2022-03-22:
**Current status:** When we initialize a planner instance, we always initialize it in 3D and we are setting the control inputs to zero in the z-direction. We did this for visualization purposes (because we need a 3D voxel map), but this is not optimal in terms of compute.

**Ideal scenario**: we would like the planner to be initialized in 2D. We may need to modify the current `planner-util` package to be able to use a slice of the 3D map (but we still can visualize it in rviz).

versatran01, 2022-03-23:
I suggest fixing the issue with inconsistent map storage order first.

XuRobotics, 2022-03-24:
> I suggest fixing the issue with inconsistent map storage order first. I don't remember if this was fixed or not but worth looking at to make sure.

I don't think this is fixed, would you like to create an issue on this?