KumarRobotics/kr_autonomous_flight #141: Fix A star error in JPS3D/graph_search.cpp

fcladera, 2022-03-22:
**Current status:** ` printf("ASTAR ERROR!\n");` is called in `grap_search.cpp`.  This does not seem to affect the performance of the stack.

**Ideal scenario:** we would like to understand why this is happening and fix it. Removing the printf is not acceptable.

orlando21lara, 2022-05-20:
Hi, I'm trying to reproduce this error, is there any known scenerios in which this occurs?

XuRobotics, 2022-05-21:
Hi Orland, It’s hard to say what exact scenario causes this. But have you tried to build the stack and run in the simulator? If not, I would recommend doing that first, and try bunch of flights in different cluttered environments (eg forests).

orlando21lara, 2022-05-21:
Hi Xu, yes I have run the simulator with the base environment which does have a quite of bit of trees. After setting a few waypoints I have only been able to see the error once, but I did not notice what trajectory or situation could have caused it. I have not been able to reproduce it, which is why I was wondering if there was any know situation in which it occured.

Is the JPS3D code in this repository based off of https://github.com/KumarRobotics/jps3d ? I git diffed the `graph_search.cpp` files in both repos and there seems to only be formatting differences. Since they both have this `printf("ASTAR ERROR!\n");`, I was thinking that I could run more repeatable test cases on this isolated repo instead to make debugging quicker. Would you recommend  I do that or advise against it?

Would you have any other recommendations?

XuRobotics, 2022-05-21:
I see. Yes that’s correct! In general, yes I think it’s a good idea use the isolated repo to identify the issues first, I am not very sure if you’ll run into this issue by running the JPS alone though. Maybe you can pick another issue to solve instead? Next time when I get the issue I will note down what exact scenario it is.

orlando21lara, 2022-05-21:
Alright, thank you for the feedback. I'll look at some other issue for now and come back to this in the future