KumarRobotics/kr_autonomous_flight #146: Reliable integration of LIDAR odometry into autonomy stack

XuRobotics, 2022-04-20:
We have tried the integration of LIDAR odometry into UKF, using [LLOL](https://github.com/versatran01/llol). We successfully flew waypoint missions, using LIDAR only for both odometry and mapping (i.e. no cameras were needed for the autonomous flight). 

However, this is still in an early stage and we have not done large-scale tests. It would be good to test and compare the odometry performance with different combinations of sensors, including (1) stereo-VIO-only (2) LIDAR-odom-only (3) stereo-VIO + LIDAR-odom. 

The interfaces to support different sensors are all available in the UKF package included in this repo.

closed 2022-08-24 by XuRobotics

nvhungv2k, 2024-06-23:
@XuRobotics You said "We successfully flew waypoint missions, using LIDAR only for both odometry and mapping (i.e. no cameras were needed for the autonomous flight)". Doing it again, following the instruction [here](https://github.com/KumarRobotics/kr_autonomous_flight/wiki/LIDAR-only-autonomous-flight), isn't it?
If TRUE, I followed it as:
- Clone & Build:
```
git clone https://github.com/KumarRobotics/kr_autonomous_flight.git `
cd kr_autonomous_flight/
git checkout feature/integrate_lidar_no_3d_planner
vcs import < external_all.yaml
vcs import < external_lidar_odometry.yaml
vcs pull
cd ../..
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build
```
- Run
```
cd /path/to/workspace
source devel/setup.bash
roslaunch gazebo_utils simulation.launch
```
I don't make any changes in source code. I did do the sequential steps: motor on - take off - publish waypoints. Finally, when I pressed "Execute waypoint mission", I always received as:
[image]
And obviously Quadrotor don't move
I guessed that it seemed no pointcloud data come from LiDAR, but otherwise, I checked LiDAR topic:
`rostopic echo /falcon4/os1_cloud_node/points`
it is good.
PLS show me why? Thanks so much!

XuRobotics, 2024-06-23:
Make sure you follow instructions here: https://github.com/KumarRobotics/kr_autonomous_flight/wiki/Gazebo-Simulation-Setup

Specifically, make sure ouster ros repo is on the simulator branch

Also you’re using the wrong launch file, it should be full_sim.launch as specified by the link above

Another additional change you can try, if you still have problems with the stimulator experiments, is to reduce map resolution (increase the voxel size in the corresponding mapper.yaml file), and make sure you did build the code in release mode, to avoid any computation related problems.

XuRobotics, 2024-06-23:
PS: FYI, I pasted the relevant command in the exteral_all.yaml that allows you to use the simulator branch of ouster ros

ouster_example:
    type: git
    url: http://github.com/XuRobotics/ouster_example.git
    version: os-sim

nvhungv2k, 2024-06-23:
> PS: FYI, I pasted the relevant command in the exteral_all.yaml that allows you to use the simulator branch of ouster ros
> 
> ouster_example: type: git url: [http://github.com/XuRobotics/ouster_example.git](https://github.com/XuRobotics/ouster_example.git) version: os-sim

I just check ouster_example as you said, it already was in os-sim branch:
[image]

XuRobotics, 2024-06-23:
By the way, we never did experiments with lidar odom in simulation, we have been using our robot with Ouster OS1-64 lidar to do these experiments directly

nvhungv2k, 2024-06-23:
> By the way, we never did experiments with lidar odom in simulation, we have been using our robot with Ouster OS1-64 lidar to do these experiments directly

I will try it in simulation.
Pls get back to kr_autonomous_fligh repo on master branch.
I ran it completely as [tutorial video](https://drive.google.com/drive/folders/1_u_9vhmQYLvpo80egivIvrqPKD28iv4K?usp=sharing) you posted. One thing I don't understand that:
1.  In default, it don't use S-MSCKF VIO (because onboard_sensing = false in [full_sim.launch](https://github.com/KumarRobotics/kr_autonomous_flight/blob/dba55a7652c9cec2557bfa43f3c2a01fe38aa4d4/autonomy_sim/gazebo_sim/gazebo_utils/launch/full_sim.launch#L19)). So which source the planner get the location from to create path and trajectory?
2. I make it used S-MSCKF VIO by setting onboard_sensing = true in [full_sim.launch](https://github.com/KumarRobotics/kr_autonomous_flight/blob/dba55a7652c9cec2557bfa43f3c2a01fe38aa4d4/autonomy_sim/gazebo_sim/gazebo_utils/launch/full_sim.launch#L19). S-MSCKF VIO published to /quadrotor/ukf_odom topic, BUT no one used it (that is, no subscriber):
[image]
So in this case which role does S-MSCKF VIO play?
I supposed that S-MSCKF VIO is as backbone for any SLAM (ex: SLOAM,...) to correct the drift.
PLS show me more explaination. Thanks so much!

nvhungv2k, 2024-06-23:
> However, this is still in an early stage and we have not done large-scale tests. It would be good to test and compare the odometry performance with different combinations of sensors, including (1) stereo-VIO-only (2) LIDAR-odom-only (3) stereo-VIO + LIDAR-odom. Also it would be good to idenfy degenerative cases for different combinations.
> 
@XuRobotics I have been doing different combinations of sensors as you said. But I am in stage of running them in gazebo simulation. So what factors do you use to compare them? Can you share it or refer to any paper related to? THANKS!