KumarRobotics/kr_autonomous_flight #149: Make the voxel map's default values for occupied, unknown and free voxels as parameters in VoxelMap.msg

XuRobotics, 2022-04-21:
Currently, the default values for occupied, even, unknown and free voxels are hard-coded, respectively, as val_occ, val_even, val_unkonwn, and val_free. The planner and mapper have a consensus on the hard-coded values. This may cause problems, e.g., when someone wants to use different planners.

A better way to do this is to, in the VoxelMap.msg, add parameters describing those default values that describe the occupancy status. 

(val_even is a value between val_free and val_unkonwn, and when the mapper decay is turned on, it is used as a threshold for determining the occupancy status).

ankit16-py, 2022-04-28:
Hey! I am trying to work on this issue and I was wondering if you can answer some doubts that I have. Thank you in advance.

1) I am fairly new to ros so I don't understand what it means to add "parameters" to a msg file. Does that mean to just extend the msg file with new variables for the val_occ, val_even, etc?

2) Where exactly are the variables hard-coded? I was hoping to look at the values to understand what type of data they store.

XuRobotics, 2022-04-28:
Hi Ankit, I will send a private msg to you.

XuRobotics, 2022-04-28:
Hi Ankit, please see below for my explanation:

1. Check the [VoxelMap.msg](https://github.com/KumarRobotics/kr_autonomous_flight/blob/d5b9a659b7e4414f6c22e5a04b1ebec42a398753/autonomy_core/map_plan/planning_ros_msgs/msg/VoxelMap.msg), currently we have the following parameters:

 `std_msgs/Header header

# Resolution 
float32 resolution

# Origin is the origin of map stored as float
geometry_msgs/Point origin
# Dim is the dimension of the map stored as int
geometry_msgs/Point dim

# Subscript (xi, yi, zi) indices probability data[xi + dim.x * yi + zi * dim.x * dim.y]
int8[] data
`

You will need to add extra parameters to describe the occupancy conditions of each voxel in the voxel map. You will need to at least add the following 2 parameters:
`
int8_t val_free = 0;
int8_t val_occ = 100;
`
Where the 1st parameter val_free represents the value of free voxels, and the 2nd parameter val_occ represents the value of occupied voxels.

2. They are hard-coded in mapper package, e,g., [here](https://github.com/KumarRobotics/kr_autonomous_flight/blob/d5b9a659b7e4414f6c22e5a04b1ebec42a398753/autonomy_core/map_plan/mapper/include/mapper/voxel_mapper.h#L158)  as well as in planner package, e.g., [here](https://github.com/KumarRobotics/kr_autonomous_flight/blob/d5b9a659b7e4414f6c22e5a04b1ebec42a398753/autonomy_core/map_plan/jps3d/include/jps/map_util.h#L111) and [here](https://github.com/KumarRobotics/kr_autonomous_flight/blob/d5b9a659b7e4414f6c22e5a04b1ebec42a398753/autonomy_core/map_plan/mpl/include/mpl_collision/map_util.h#L89). 

You need to find all of them, remove the hard coding, and load them from the VoxelMap.msg.

Feel free to follow up if there's still any confusion!

ankit16-py, 2022-05-02:
Hey! So I added the parameters to the message file and it successfully build. But when I try to use the VoxelMap.h file in map_utils.h (`#include <planning_ros_msgs/VoxelMap.h>`) to swap out the hard coded values it gives me an error while building the package (it points to the include statement above and says no such file or directory present). Is this a package dependency error because I edited the CMakeLists and packages.xml of the jps3d package (where map_utils.h resides) and it still gave me the same error. I am attaching the code below for the 2 files.

CMakeLists.txt of jps3d

```
cmake_minimum_required(VERSION 3.10)
project(jps3d)

set(CMAKE_CXX_STANDARD 17)

find_package(catkin REQUIRED
            COMPONENTS planning_ros_msgs)
find_package(Eigen3 REQUIRED)
find_package(Boost REQUIRED)

catkin_package(INCLUDE_DIRS include LIBRARIES ${PROJECT_NAME})

add_library(${PROJECT_NAME} src/graph_search.cpp src/jps_planner.cpp
                            src/map_util.cpp)
target_include_directories(${PROJECT_NAME} PUBLIC include)
target_link_libraries(${PROJECT_NAME} PUBLIC Eigen3::Eigen Boost::boost)
```
packages.xml of jps3d

```
<?xml version="1.0"?>
<package format="2">
  <name>jps3d</name>
  <version>1.0.0</version>
  <description>The jps3d package</description>
  <maintainer email="sikang@seas.upenn.edu">sikang</maintainer>

  <license>BSD-3-Clause</license>
  <buildtool_depend>catkin</buildtool_depend>

  <depend>planning_ros_msgs</depend>

  <export>
  </export>

</package>
```

XuRobotics, 2022-05-02:
Hi Ankit, would you mind pasting your error message here? Also, did you make sure that you include the folder of VoxelMap.msg for all targets that use the VoxelMap message?

ankit16-py, 2022-05-02:
```
Errors     << jps3d:make /home/ankit/Work/Projects/kraf/logs/jps3d/build.make.017.log
In file included from /home/ankit/Work/Projects/kraf/src/kr_autonomous_flight/autonomy_core/map_plan/jps3d/src/map_util.cpp:1:
/home/ankit/Work/Projects/kraf/src/kr_autonomous_flight/autonomy_core/map_plan/jps3d/include/jps/map_util.h:8:10: fatal error: planning_ros_msgs/msg/VoxelMap.h: No such file or directory
    8 | #include <planning_ros_msgs/msg/VoxelMap.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/jps3d.dir/build.make:89: CMakeFiles/jps3d.dir/src/map_util.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ankit/Work/Projects/kraf/src/kr_autonomous_flight/autonomy_core/map_plan/jps3d/include/jps/jps_planner.h:12,
                 from /home/ankit/Work/Projects/kraf/src/kr_autonomous_flight/autonomy_core/map_plan/jps3d/src/jps_planner.cpp:1:
/home/ankit/Work/Projects/kraf/src/kr_autonomous_flight/autonomy_core/map_plan/jps3d/include/jps/map_util.h:8:10: fatal error: planning_ros_msgs/msg/VoxelMap.h: No such file or directory
    8 | #include <planning_ros_msgs/msg/VoxelMap.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/jps3d.dir/build.make:76: CMakeFiles/jps3d.dir/src/jps_planner.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:544: CMakeFiles/jps3d.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/ankit/Work/Projects/kraf/build/jps3d; catkin build --get-env jps3d | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

```
Above is the error that happen when I try to build the workspace. Also I don't understand what it means to "include the folder" of VoxelMap.msg. Thanks.

XuRobotics, 2022-05-02:
target_include_directories https://cmake.org/cmake/help/latest/command/target_include_directories.html

ankit16-py, 2022-05-02:
Thanks for the information! I was able to build successfully now with having removed hardcodings from the two map_utils.h files and one voxel_mapper.h file. I also checked to make sure no other files had the hard-codings for the said variables. I then ran the gazebo simulator to make sure everything was working properly and the quadrotor was able to navigate the forest map properly as before. Is there any other way to make sure I have done things properly?

XuRobotics, 2022-05-02:
Great job! Could you please create a Pull Request for this? After you create that, I will take a look. Thanks!

ankit16-py, 2022-05-02:
Yes I will do it asap! I just need to lookup the proper way to do this.

ankit16-py, 2022-05-02:
Hey! I just created a PR with the latest changes. As instructed, I also added the val_add and val_default parameters to the msg file and replaced the hardcodings.

XuRobotics, 2022-05-13:
UPDATE: Changing label to hard, we need to further improve this according to Chao's suggestion here:

https://github.com/KumarRobotics/kr_autonomous_flight/pull/151#issuecomment-1115543192

ankit16-py, 2022-05-13:
Ok! I will look into making this improvement!