KumarRobotics/kr_autonomous_flight #177: [quadrotor/rqt_gui_buttons-22] process has died

yhy130531, 2022-09-12:
Thanks for open source your excellent work, I got the following error while executing "roslaunch gazebo_utils full_sim.launch":

[quadrotor/rqt_gui_buttons-22] process has died [pid 25168, exit code 1, cmd /opt/ros/melodic/lib/rqt_gui/rqt_gui --perspective-file /home/hy/kr_UAV_ws/src/kr_autonomous_flight/autonomy_core/client /client_launch/config/client_gui.perspective robot:=quadrotor __name:=rqt_gui_buttons __log:=/home/hy/.ros/log/e77eaff4-326f-11ed-8883-9c2976e53ddc/quadrotor-rqt_gui_buttons-22.log].
log file: /home/hy/.ros/log/e77eaff4-326f-11ed-8883-9c2976e53ddc/quadrotor-rqt_gui_buttons-22*.log

How to solve it?

[image]

yhy130531, 2022-10-11:
> I don't remember seeing this error before. It could be related to the dependency issues or how you build this repository.
> 
> Remember that you need to change all "noetic" to "melodic" if you're using ubuntu 18.04 + ROS melodic when you [build from source](https://github.com/KumarRobotics/kr_autonomous_flight/wiki/Building-from-Source)
> 
> Our code should work on 18.04 + ROS melodic. But I would recommend trying [using docker](https://github.com/KumarRobotics/kr_autonomous_flight/wiki/Using-Docker), if you cannot get it to work.

Thank you, this problem has been solved!

closed 2022-10-11 by yhy130531

zzhh00, 2023-01-29:
@yhy130531 HI, I meet the same porblem, can you share your solution please? Thank you.

david-yan, 2023-02-09:
I had this problem as well. `rqt` did not work. Was able to resolve by uninstall `PyQt5` and `PyQt5-sip` with pip3. 

https://answers.ros.org/question/371691/valueerror-pycapsule_getpointer-called-with-incorrect-name/