KumarRobotics/kr_autonomous_flight #183: gazebo simulation waypoints in different z axis

15707397862, 2022-12-09:
I tried to set some wappoints of different heights in the my gazebo world, but it seems that the UAV cannot move on z axis. Can someone help me.
[image]

XuRobotics, 2022-12-16:
The default mode is 2D planning. You will need to do the following to enable 3D planning mode

Open the file tracker_params_mp.yaml (https://github.com/KumarRobotics/kr_autonomous_flight/blob/master/autonomy_core/control/control_launch/config/tracker_params_mp.yaml).

You will need to set **use_3d_local** and **use_3d_global** to true. 

You can modify **vertical_semi_fov** parameter so that the UAV always plans in the sensor vertical field of view to guarantee safety (but may result in more failures due to limited motion primitives). 

We are developing a new more capable and efficient 3D planning algorithm that will soon be merged back to the master branch, stay tuned!

15707397862, 2022-12-17:
thank u sir, and i will try it late. i hope you are in good health at this special time.

closed 2023-02-19 by XuRobotics