KumarRobotics/kr_autonomous_flight #184: how to replace lidar with realsense depth camera

panchal-harsh, 2022-12-16:


XuRobotics, 2022-12-16:
Are you running with a real robot? Or in gazebo simulation? Either way, two things you will need to change:

**1. point cloud topic, change this to whatever is published by realsense.**
For gazebo, see here: https://github.com/KumarRobotics/kr_autonomous_flight/blob/9d9a87eb84ff47b77242cd457d367001d5b8ac49/autonomy_sim/gazebo_sim/gazebo_utils/launch/full_sim.launch#L12
For real robot, see here: https://github.com/KumarRobotics/kr_autonomous_flight/blob/9d9a87eb84ff47b77242cd457d367001d5b8ac49/autonomy_real/real_experiment_launch/launch/full_autonomy.launch#L65

**2. rigid body transform from point cloud frame (realsense) to robot body frame.** 
For simulation, it should directly work since Gazebo publish the static tf using the robot URDF you specified. 
For real robot, see here: 
https://github.com/KumarRobotics/kr_autonomous_flight/blob/9d9a87eb84ff47b77242cd457d367001d5b8ac49/autonomy_real/real_experiment_launch/launch/publish_tf.launch#L14

XuRobotics, 2022-12-16:
One thing to note is that our current planner assumes the sensor has 360 degrees horizontal field of view (like LiDAR/Omni directional cameras). 

Make sure that you set align_yaw to be true (see the yaml file linked below), and maybe set yaw velocity to a higher value so that the UAV aligns the heading with its movement direction fast enough to see obstacles. 

Soon in the future we will integrate perception aware planning algorithm into the stack to guarantee safety for sensors with limited field of view.

tracker_params_mp.yaml (https://github.com/KumarRobotics/kr_autonomous_flight/blob/master/autonomy_core/control/control_launch/config/tracker_params_mp.yaml).

panchal-harsh, 2022-12-23:
Thanks for your response ,I am trying to run with real robot ,In this setup an external IMU is used but can I just use the pixhawk IMU along with intel realsense's IMU will the accuracy still be satisfactory and where the changes need to be done in the code for that , I just want to use an Intel realsense ,pixhawk 4 ,jetson nano and companian computer to perform autonomous flight is it possible with this stack and where must the changes  be done for that

closed 2023-02-09 by XuRobotics