KumarRobotics/kr_autonomous_flight #185: Imu quality

poudyalbot, 2023-01-06:
Res sir,
What is the minimum imu quality or imu type required for realtime control of robot using vio because vn imu are very costly and can it work with imu used on smartphone.
Thank you

XuRobotics, 2023-01-06:
Hi, For experiments presented in the papers linked in this repo, we only tried high-quality IMUs such as vectornav VN-100. For the control loop to stay stable, I would recommend you use odometry with enough frequency (usually >= 100 Hz). We have other projects that use cellphone-grade sensors and processors for autonomous flight, such as [this one](https://www.grasp.upenn.edu/projects/smartphones-power-flying-robots/). A quick search can lead you to open-source VIO repositories that support cellphone-grade sensors, such as [VINS-Mono](https://github.com/HKUST-Aerial-Robotics/VINS-Mono).

poudyalbot, 2023-01-07:
Thanks for the reply.

closed 2023-02-09 by XuRobotics