KumarRobotics/kr_autonomous_flight #190: protobuf issue - Errors     << ouster_gazebo_plugins:check /home/minasm/catkin_ws/logs/ouster_gazebo_plugins/build.check.010.log

suvasis, 2024-02-14:
hi,

How to fix build issue:

protoc --version
libprotoc 3.21.12

catkin build throws the following error:

Errors     << ouster_gazebo_plugins:check /home/minasm/catkin_ws/logs/ouster_gazebo_plugins/build.check.010.log
CMake Warning (dev) at CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

catkin build
-----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/minasm/catkin_ws
-----------------------------------------------------------
Build Space:        [exists] /home/minasm/catkin_ws/build
Devel Space:        [exists] /home/minasm/catkin_ws/devel
Install Space:      [unused] /home/minasm/catkin_ws/install
Log Space:          [exists] /home/minasm/catkin_ws/logs
Source Space:       [exists] /home/minasm/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found 50 packages in 0.0 seconds.                                                            
[build] Package table is up to date.                                                                 
Starting  >>> client_launch                                                                          
Starting  >>> control_launch                                                                         
Starting  >>> estimation_launch                                                                      
Starting  >>> fla_ukf                                                                                
Starting  >>> gazebo_utils                                                                           
Starting  >>> kr_mav_msgs                                                                            
Starting  >>> kr_mesh_visualization                                                                  
Starting  >>> kr_planning_msgs                                                                       
Starting  >>> kr_tracker_msgs                                                                        
Starting  >>> map_plan_launch                                                                        
Starting  >>> mrsl_models                                                                            
Starting  >>> mrsl_quadrotor_description                                                             
Starting  >>> mrsl_quadrotor_utils                                                                   
Starting  >>> msckf_vio                                                                              
Starting  >>> ouster_client                                                                          
Starting  >>> ouster_description                                                                     
Starting  >>> ouster_gazebo_plugins                                                                  
Starting  >>> px4_interface_launch                                                                   
Starting  >>> real_experiment_launch                                                                 
Starting  >>> state_machine_launch                                                                   
Starting  >>> waypoint_navigation_plugin                                                             
Finished  <<< control_launch                            [ 0.1 seconds ]                              
Finished  <<< ouster_description                        [ 0.1 seconds ]                              
Finished  <<< estimation_launch                         [ 0.1 seconds ]                              
Finished  <<< real_experiment_launch                    [ 0.1 seconds ]                              
Finished  <<< fla_ukf                                   [ 0.2 seconds ]                              
Finished  <<< state_machine_launch                      [ 0.1 seconds ]                              
Finished  <<< kr_tracker_msgs                           [ 0.3 seconds ]                              
Finished  <<< map_plan_launch                           [ 0.1 seconds ]

fcladera, 2024-02-14:
Please post a clear log of the issue using [Markdown formatting for code](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code).  I am not able to see what the problem is from your log.

The current version of the code was built [4 days ago without issues](https://github.com/KumarRobotics/kr_autonomous_flight/actions).

suvasis, 2024-02-14:
It's complaining about much newer version of protoc --version libprotoc 3.21.12. What is the protobuf version for you compilation?

suvasis, 2024-02-14:
Errors     << ouster_gazebo_plugins:make /home/minasm/catkin_ws/logs/ouster_gazebo_plugins/build.make.008.log
In file included from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/minasm/catkin_ws/src/kr_autonomous_flight/ouster_example/ouster_gazebo_plugins/include/ouster_gazebo_plugins/GazeboRosOusterLaser.h:49,
                 from /home/minasm/catkin_ws/src/kr_autonomous_flight/ouster_example/ouster_gazebo_plugins/src/GazeboRosOusterLaser.cpp:35:
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/minasm/catkin_ws/src/kr_autonomous_flight/ouster_example/ouster_gazebo_plugins/include/ouster_gazebo_plugins/GazeboRosOusterLaser.h:49,
                 from /home/minasm/catkin_ws/src/kr_autonomous_flight/ouster_example/ouster_gazebo_plugins/src/GazeboRosOusterLaser.cpp:35:
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/physics/Model.hh:28,
                 from /usr/include/gazebo-11/gazebo/physics/Actor.hh:30,
                 from /usr/include/gazebo-11/gazebo/physics/physics.hh:2,
                 from /home/minasm/catkin_ws/src/kr_autonomous_flight/ouster_example/ouster_gazebo_plugins/include/ouster_gazebo_plugins/GazeboRosOusterLaser.h:49,
                 from /home/minasm/catkin_ws/src/kr_autonomous_flight/ouster_example/ouster_gazebo_plugins/src/GazeboRosOusterLaser.cpp:35:
/usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:25:10: fatal error: google/protobuf/generated_message_table_driven.h: No such file or directory

fcladera, 2024-02-14:
Alright, I think that I know what may be happening. Let me rebuild the docker image and I will confirm.

fcladera, 2024-02-14:
Just to confirm, are you using the right version of  `ouster_ros` for simulation? Our sim image is building without any issues: https://github.com/KumarRobotics/kr_autonomous_flight/actions/runs/7908421589/job/21587556958

suvasis, 2024-02-14:
I am following Building from Source step by step. Is there anything I need to do for ouster_ros? I am not using docker for now.

suvasis, 2024-02-15:
I checked docker instructions. It's working perfectly. So the "build from source" is not working.

fcladera, 2024-02-15:
Thanks. This is probably a dependency that has not been tagged to the right version.

suvasis, 2024-02-15:
Please let me know when you fix this issue.

suvasis, 2024-02-17:
hi,
Is there any update on this?