KumarRobotics/kr_autonomous_flight #192: Can't simulate with forest_Large

nvhungv2k, 2024-06-06:
Hello everyone,
I am having problems when I changed default environment 'forest0' to 'forest_Large'.
I did the same steps as for both environment. But with 'forest_Large', UAV didn't move and shown the following information:
[image]

closed 2024-06-07 by nvhungv2k

fcladera, 2024-06-07:
Hi @nvhungv2k, I am glad you figured this out! Would it be possible to document it here for future people facing this problem?

nvhungv2k, 2024-06-12:
I figured that
With large environment, the consumed time to launch is higher. I pressed "motor on" and "take off" so early when the system does not init completely.
So I should wait until I received the notification that "State machine transitioning 'IdleTransition':'succeeded'-->'Off'"