KumarRobotics/kr_autonomous_flight #194: Accumulative DRIFT when using only-VIO in gazebo simulation

nvhungv2k, 2024-06-24:
Hi guys,
I am trying to run with only VIO in gazebo simulation
Firstly, I followed [this](https://github.com/KumarRobotics/kr_autonomous_flight/wiki/Building-from-Source) to build and then [to simulate in gazebo](https://github.com/KumarRobotics/kr_autonomous_flight/wiki/Gazebo-Simulation-Setup).
I figured out that the odometry which is for planning task come from gazebo directly. So I want to replace Gazebo's odometry by VIO's odometry (particularly in this case it is MSCKF-VIO)
First and foremost, I need to get odometry from MSCKF-VIO and check it. I did do as following:
1. Feeding IMU and CAMERA from gazebo to MSCKF-VIO: edit [system_mp.launch](https://github.com/KumarRobotics/kr_autonomous_flight/blob/dba55a7652c9cec2557bfa43f3c2a01fe38aa4d4/autonomy_core/state_machine/state_machine_launch/launch/system_mp.launch#L23)
```
<arg name="cam0" default="cam_left"/>
<arg name="cam1" default="cam_right"/>
 <arg name="imu" default="imu"/>
```
-->
```
<arg name="cam0" default="/quadrotor/ovc3/left"/>
<arg name="cam1" default="/quadrotor/ovc3/right"/>
<arg name="imu" default="/falcon4/os1_cloud_node/imu"/>
```
2. Change the calibration file in [system_mp.launch](https://github.com/KumarRobotics/kr_autonomous_flight/blob/dba55a7652c9cec2557bfa43f3c2a01fe38aa4d4/autonomy_core/state_machine/state_machine_launch/launch/system_mp.launch#L29):
`<arg name="calibration_file" default="msckf_calib.yaml"/>`
-->
`<arg name="calibration_file" default="$(find estimation_launch)/config/msckf_calib_simulation.yaml"/>`
3. Change image topics in calibration file: [msckf_calib_simulation.yaml](https://github.com/KumarRobotics/kr_autonomous_flight/blob/dba55a7652c9cec2557bfa43f3c2a01fe38aa4d4/autonomy_core/estimation/estimation_launch/config/msckf_calib_simulation.yaml#L14)
`cam0: rostopic: /quadrotor/cam_left/image_raw ` --> `cam0: rostopic: /quadrotor/ovc3/left/image_raw `
`cam1: rostopic: /quadrotor/cam_right/image_raw`  --> `cam1: rostopic: /quadrotor/ovc3/right/image_raw `
Finally, I run and view odom from MSCKF (via /quadrotor/vio/odom and /quadrotor/ukf_odom topics) on RVIZ
Run: `roslaunch gazebo_utils full_sim.launch` (gazebo word is still default, it is [forest0](https://github.com/KumarRobotics/kr_autonomous_flight/blob/dba55a7652c9cec2557bfa43f3c2a01fe38aa4d4/autonomy_sim/gazebo_sim/gazebo_utils/launch/full_sim.launch#L35C36-L35C43))
View on Rviz:
[image]
In above GIF: fix frame is truth odom form gazebo, two remaining frames is odom from MSCKF-VIO
We saw that odom's DRIFT from MSCKF-VIO is accumulated.

XuRobotics, 2024-06-24:
As stated in our paper and our wiki, we use high fidelity Unity-based simulator for VIO simulation experiments. In gazebo, we use ground truth Odom since its image quality is not enough for running the vio.

nvhungv2k, 2024-06-24:
> As stated in our paper and our wiki, we use high fidelity Unity-based simulator for VIO simulation experiments. In gazebo, we use ground truth Odom since its image quality is not enough for running the vio.

As you said. Couldn't we also use any vision-based odometry in gazebo simulation?
So what about the LiDAR-ONLY-based odometry such as [LLOL](https://github.com/versatran01/llol), [LOAM](https://github.com/HKUST-Aerial-Robotics/A-LOAM),...?

closed 2024-07-23 by XuRobotics