KumarRobotics/kr_autonomous_flight #195: TAKE OFF ERROR when only using Fast-LIO Odometry

nvhungv2k, 2024-06-26:
Hi everyone,
I am trying to test kr_autonomous_flight with only lidar odometry (particularly fast-LIO) in gazebo simulation (default environment is forest0)
I tested and compared the result with truth odometry getting from gazebo directly. it is quitly good. Pls view [this video](https://youtu.be/siLgkC6z1NM)
But when using Fast-LIO odometry for mapping and planning, the error comes from the step of TAKEOF as soon as. The Quadrotor falled freely and then tried to take off. Finally final altitude is not good. Pls view [this video](https://youtu.be/qfcW9cAM2Rk)
Any ideas. Pls support me. Thanks!

XuRobotics, 2024-07-23:
Hi Nguyen,

The odometry seems to be drifting in the simulator, and I'm not sure how well Fast-LIO works with the Gazebo point cloud data. There may be some simulation artifacts causing the odometry algorithm to fail. On the autonomy stack side, if the odometry becomes unstable, there is very little that can be done to alleviate the problem.

One thing you can try is to tune the SE3 controller gains. However, the critical issue remains to ensure that your odometry is stable.

XuRobotics, 2024-07-23:
Another idea is to try environments with better vertical structures. We included some other environments in our repository; for example, [the pillars environment](https://github.com/KumarRobotics/mrsl_quadrotor/blob/0cad700ea2d67711f3a961398dbb80b74026be7a/mrsl_quadrotor_description/worlds/pillars.world) may be a good option. Alternatively, you can build a simple environment yourself or download one from online asset websites that has good vertical structures to help lidar odometry perform better.

Additionally, another lidar odometry algorithm worth trying is Faster LIO (https://github.com/gaoxiang12/faster-lio), which we have been using on our robot for over a year. It is considered an improved version of Fast-LIO2 and may provide better stability and performance.

closed 2024-07-23 by XuRobotics